awawa-dev/HyperSerialPico #1: Raspberry Pi Pico "Reset" in Device Manager appears after flash

KenshiHH, 2023-05-09:
hey,

just flashed the "HyperSerialPico_sk6812Cold.uf2" onto a Raspberry Pi Pico,

a "Reset" Device appeared after the Flash in the Device Manager under "other"

[image]

Device itself gets Recognized in HyperHDR v20beta

`2023-05-09T13:08:18.388Z [LEDDEVICE_ADALIGHT] Opening UART: COM9
2023-05-09T13:08:18.388Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:216) _rs232Port.open(QIODevice::ReadWrite): COM9, Baud rate [2000000]bps
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:221) portName:          COM9
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:222) systemLocation:    \\.\COM9
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:223) description:       Serielles USB-Ger�t
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:224) manufacturer:      Microsoft
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:225) productIdentifier: 0xa
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:226) vendorIdentifier:  0x2e8a
2023-05-09T13:08:18.390Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:227) serialNumber:      E66138935F287729
2023-05-09T13:08:18.415Z [LEDDEVICE_ADALIGHT] Detected Rp2040 type board. HyperHDR skips the reset. State: 0, 0
2023-05-09T13:08:18.496Z [LEDDEVICE_ADALIGHT] ESP sent: 'HyperHDR frames: 0 (FPS), receiv.: 0, good: 0, incompl.: 0, mem1: 0, mem2: 206, heap: 124256'
2023-05-09T13:08:18.496Z [LEDDEVICE_ADALIGHT] ESP sent: 'RGBW => Gain: 255/255, red: 160, green: 160, blue: 160'
2023-05-09T13:08:18.496Z [LEDDEVICE_ADALIGHT] DETECTED DEVICE USING HYPERSERIALESP8266/HYPERSERIALESP32 FIRMWARE (Welcome! Awa driver 9.) at 1 msec`

awawa-dev, 2023-05-09:
Hi
You don't need worry about it, 'Reset device' is used by some Raspberry Pico developer tools https://forums.raspberrypi.com/viewtopic.php?t=324909 . It won't interfere with HyperSerialPico (Pico COM9 is correctly detected in the device manager).

KenshiHH, 2023-05-09:
Alright, thx for the quick response :)
Have a nice day

closed 2023-05-09 by KenshiHH

awawa-dev, 2023-05-09:
Because I see that you built your version of HyperSerialPico using Github Action you can add
`add_definitions ( -DPICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE=0 )`
at the top of the CMakeList.txt and re-compile it. `Reset device` will disappear after the new firmware is uploaded.

KenshiHH, 2023-05-09:
It worked :)

Thanks!