awawa-dev/HyperSerialPico #5: Raspberry PI Pico W not detected

RaidMax, 2023-06-16:
Trying this out on Windows.
Followed instructions for flashing and configuration.
Does not seem to detect the device via the COM Port.

```
2023-06-16T20:39:15.310Z [LEDDEVICE_ADALIGHT] Opening UART: COM4
2023-06-16T20:39:15.310Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:216) _rs232Port.open(QIODevice::ReadWrite): COM4, Baud rate [200000]bps
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:221) portName:          COM4
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:222) systemLocation:    \\.\COM4
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:223) description:       USB Serial Device
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:224) manufacturer:      Microsoft
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:225) productIdentifier: 0xa
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:226) vendorIdentifier:  0x2e8a
2023-06-16T20:39:15.311Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:227) serialNumber:      E6616408432Fxxxx
2023-06-16T20:39:16.630Z [LEDDEVICE_ADALIGHT] Could not detect HyperSerialEsp8266/HyperSerialESP32 device
2023-06-16T20:39:16.630Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:333) Switch on
2023-06-16T20:39:16.630Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:409) Power On
2023-06-16T20:39:16.631Z [COMPONENTREG0] LED device: enabled
2023-06-16T20:39:17.633Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:300) Timeout after 1000ms: 0 frames already dropped
2023-06-16T20:39:18.634Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:300) Timeout after 1000ms: 1 frames already dropped
```

awawa-dev, 2023-06-16:
Are you sure you are using HyperHDR v20 beta with 'Esp8266/ESP32/**Rp2040** handshake' enabled?

RaidMax, 2023-06-16:
Let me try with the beta. I saw that the requirement was added 5 days ago in the readme corresponding to the last release, so I thought it was only for the latest release. Is there a v20 binary release or does it require compilation. I can't seem to find it on the HyperHDR releases.

awawa-dev, 2023-06-16:
The last commit only clarified this as the earlier version of HyperHDR didn't have "Esp8266/ESP32/Rp2040 handshake" (v20), only "Esp8266/ESP32 handshake" (v19). Currently, the latest beta installers can be found in the build artifacts here: https://github.com/awawa-dev/HyperHDR/actions/runs/5249279247

RaidMax, 2023-06-16:
That did the trick thanks. Sorry for the false alarm.

closed 2023-06-16 by RaidMax