awawa-dev/HyperSerialPico #22: Fast fast can these pico go...

Bascht74, 2024-03-17:
Hi, 

is the max. baud rate 2Mbit/s oder can you go higher with these picos?
I switched from my ESP8266 and I did not change this value, but I am wondering if I should...

I am using the 2040 with level shift :-)

Mayby you could add that information in the readme.md as well :-)

awawa-dev, 2024-03-17:
Hi
Unlike the classic physical UART, Pico uses a virtual CDC controller that handles the speed itself. And there is a note in the read.me about that. Do a benchmark test and compare the result with my results obtained with a similar number of LEDs. If they are similar, you don't need to do anything. For me it always reaches maximum speed on the default settings.

closed 2024-03-17 by awawa-dev

Bascht74, 2024-03-17:
Ups, you are so absolutly right. I don't know why I missed that.
Thx!