awawa-dev/HyperSerialPico #25: SPI with Adafruit ItsyBitsy RP2040?

satgit62, 2024-05-28:
Hello,
Would “HyperSerialPico_Spi.uf2” work on “Adafruit ItsyBitsy RP2040”, with the DATA on GPIO 3 and CLOCK on GPIO 2 with the SPI WS2801 LEDs?
[Adafruit ItsyBitsy RP2040 pinout.pdf](https://github.com/awawa-dev/HyperSerialPico/files/15474560/Adafruit.ItsyBitsy.RP2040.pinout.pdf)

awawa-dev, 2024-05-28:
Hi
Not sure, the sk9822/apa102 protocol is not quite the same as ws2801. But they are very similar and there are using similar speed. But GPIO2/3 does not have level shifter support on this board. HyperSerialPico doesnt depend on any external library to render LEDs and currently there is no option to compile it strictly for ws2801: this protocol is not implemented yet.

satgit62, 2024-05-28:
Thank you for the prompt reply. Okay, I was afraid that the LEDs would not use the same protocol. I know that GPIO 2 and 3 do not have level converters.  
I will test when I get a chance and see what comes up. Furthermore, I can quickly solder a level shifter in between and will report back if the WS2801 behave like the APA 102. It's worth a try.