awawa-dev/HyperSerialPico #32: Update workflow to node20

AbbieDoobie, 2024-07-20:
I forked and attempted to build my own version of HyperSerialPico following the guide, but the action failed with this error:
"The following actions uses Node.js version which is deprecated and will be forced to run on node20: actions/checkout@v3. For more info: https://github.blog/changelog/2024-03-07-github-actions-all-actions-will-run-on-node20-instead-of-node16-by-default/"

From what I can gather, the master workflow needs to be updated to actions/checkout@V4. I was going to try doing this myself in the fork, but I don't know if there are any potential issues that I wouldn't realize with this change.

awawa-dev, 2024-07-20:
Hi
Your workflow failed because you made a mistake in the configuration file:
[image]

> "The following actions uses Node.js version which is deprecated and will be forced to run on node20: actions/checkout@v3. For more info"

this is just a warning, it does not prevent compilation. But yes, it's worth updating.

AbbieDoobie, 2024-07-22:
Ah, the documentation mentioned that when multi-segment is enabled the second segment is always the next GPIO pin. So I didn't even think an actual pin # was needed here, although I was confused by the name of the variable. That makes more sense, I'll assume I should use 3 for this then to match the expectation.

awawa-dev, 2024-10-23:
BTW the workflow has been updated to node20

closed 2024-10-23 by awawa-dev