awawa-dev/HyperSerialPico #40: Hyperserialpico freeze

grssll, 2024-10-26:
Adafruit Scorpio 2040 freezing after couple minutes.
Tried generic adalight driver and its working without any problems.
Logs:
```
2024-10-26T11:36:19.842Z [INSTANCE_CFG0] Config Fix: [root].videoGrabber.redSignalThreshold: Create property: redSignalThreshold with value: 5
2024-10-26T11:36:19.842Z [INSTANCE_CFG0] Config Fix: [root].videoGrabber.sDHOffsetMax: Create property: sDHOffsetMax with value: 0.75
2024-10-26T11:36:19.842Z [INSTANCE_CFG0] Config Fix: [root].videoGrabber.sDHOffsetMin: Create property: sDHOffsetMin with value: 0.25
2024-10-26T11:36:19.842Z [INSTANCE_CFG0] Config Fix: [root].videoGrabber.sDVOffsetMax: Create property: sDVOffsetMax with value: 0.75
2024-10-26T11:36:19.842Z [INSTANCE_CFG0] Config Fix: [root].videoGrabber.sDVOffsetMin: Create property: sDVOffsetMin with value: 0.25
2024-10-26T11:36:19.859Z [SMOOTHING0] High resolution clock is NOT STEADY!
2024-10-26T11:36:19.859Z [SMOOTHING0] Creating config (0) => type: Alternative, dirMode: false, pause: false, settlingTime: 150ms, interval: 20ms (50Hz), antiFlickTres: 32, antiFlickStep: 2, antiFlickTime: 250
2024-10-26T11:36:19.859Z [SMOOTHING0] Clearing queued colors before: enabling. Smoothing configuration changed: restarting timer.
2024-10-26T11:36:19.859Z [SMOOTHING0] Smoothing queue is cleared
2024-10-26T11:36:19.859Z [SMOOTHING0] Selecting config (0) => type: Alternative, directMode: false, pause: false, settlingTime: 150ms, interval: 20ms (50Hz), antiFlickTres: 32, antiFlickStep: 2, antiFlickTime: 250
2024-10-26T11:36:19.859Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2024-10-26T11:36:19.859Z [LEDDEVICE0_ADALIGHT] Ignoring user LED refresh rate. Forcing smoothing refresh rate = 50.00 Hz
2024-10-26T11:36:19.859Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:423) Refresh rate = 50.00 Hz
2024-10-26T11:36:19.859Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:367) Starting timer with interval = 20ms
2024-10-26T11:36:19.860Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:433) Refresh interval updated to 20ms
2024-10-26T11:36:19.860Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:448) LED refresh interval adjustment caused by smoothing configuration change to 20ms (proposed: 20ms)
2024-10-26T11:36:19.861Z [SMOOTHING0] Using alternative smoothing input (0)
2024-10-26T11:36:19.900Z [SMOOTHING0] Using alternative smoothing procedure (0)
2024-10-26T11:37:47.685Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000], [LED0: FPS = 23.74, send = 1425, processed = 1425, dropped = 1574]
2024-10-26T11:37:48.918Z [WEBSOCKET] (HyperAPI.cpp:1109) message: [{"command":"leddevice","ledDeviceType":"","params":{},"subcommand":"hasLedClock","tan":1}]
2024-10-26T11:38:02.985Z [WEBSOCKET] (HyperAPI.cpp:1109) message: [{"command":"leddevice","ledDeviceType":"","params":{},"subcommand":"hasLedClock","tan":1}]
2024-10-26T11:38:47.702Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.98, processed = 2999], [LED0: FPS = 23.74, send = 1425, processed = 1425, dropped = 1574]
2024-10-26T11:39:27.163Z [IMAGETOLED_MNG0] Total index number is: 28796 (memory: 28796). User sparse processing is: disabled, image size: 720 x 576, area number: 160
2024-10-26T11:39:41.627Z [IMAGETOLED_MNG0] Total index number is: 33120 (memory: 33120). User sparse processing is: disabled, image size: 720 x 576, area number: 160
2024-10-26T11:39:47.685Z [PERFORMANCE] [USB: FPS = 49.68, decoding = 2ms, frames = 2981, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.68, processed = 2980], [LED0: FPS = 23.71, send = 1422, processed = 1422, dropped = 1576]
2024-10-26T11:40:16.605Z [WEBSOCKET] (WebSocketClient.cpp:39) New connection from ::ffff:192.168.0.133
2024-10-26T11:40:23.739Z [WEBSOCKET] (HyperAPI.cpp:1109) message: [{"command":"leddevice","ledDeviceType":"adalight","subcommand":"discover","tan":574}]
2024-10-26T11:40:23.784Z [WEBSOCKET] (HyperAPI.cpp:1133) response: [{"devices":[{"name":"Auto","value":"auto"},{"name":"USB Serial (/dev/ttyUSB0)","value":"ttyUSB0"}],"ledDeviceType":"adalight"}]
2024-10-26T11:40:47.696Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000], [LED0: FPS = 23.74, send = 1425, processed = 1425, dropped = 1574]
2024-10-26T11:40:58.784Z [NET_DISCOVERY] Removing not responding WLED at 192.168.0.188:80 (wled-be5756.local)
2024-10-26T11:40:58.884Z [NET_DISCOVERY] Found WLED at 192.168.0.188:80 (wled-be5756.local)
2024-10-26T11:41:21.727Z [WEBSOCKET] (WebSocketClient.cpp:188) send close: 1000 
2024-10-26T11:41:36.970Z [SMOOTHING0] Clearing queued colors before: disabling
2024-10-26T11:41:36.971Z [SMOOTHING0] Smoothing queue is cleared
2024-10-26T11:41:36.971Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:297) Disable the device
2024-10-26T11:41:36.971Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2024-10-26T11:41:36.971Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:90) Switch off
2024-10-26T11:41:36.971Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:544) Set LED strip to black/power off
2024-10-26T11:41:36.971Z [SMOOTHING0] Using alternative smoothing input (0)
2024-10-26T11:41:37.098Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:146) Flush was successful
2024-10-26T11:41:37.850Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:164) Serial port is closed: ttyUSB0
2024-10-26T11:41:37.850Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T11:41:37.851Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:41:47.680Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:42:47.686Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:43:47.671Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 2999]
2024-10-26T11:44:47.677Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:45:47.683Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:46:47.686Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.99, processed = 2999]
2024-10-26T11:47:47.675Z [PERFORMANCE] [USB: FPS = 49.99, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.99, processed = 3000]
2024-10-26T11:48:47.681Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:49:47.685Z [PERFORMANCE] [USB: FPS = 49.99, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.99, processed = 2999]
2024-10-26T11:50:47.674Z [PERFORMANCE] [USB: FPS = 49.99, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:51:47.680Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:52:47.686Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:53:47.672Z [PERFORMANCE] [USB: FPS = 49.99, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.99, processed = 2999]
2024-10-26T11:54:47.678Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.99, processed = 3000]
2024-10-26T11:55:47.684Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T11:56:47.670Z [PERFORMANCE] [USB: FPS = 49.99, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 49.99, processed = 2999]
2024-10-26T11:57:47.676Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
[... cut ...]
2024-10-26T17:39:41.251Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T17:39:41.251Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T17:39:47.684Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 2999]
2024-10-26T17:40:04.216Z [WEBSOCKET] (HyperAPI.cpp:957) log streaming activated for client ::ffff:192.168.0.133
```

grssll, 2024-10-26:
```
024-10-26T17:37:42.904Z [IMAGETOLED_MNG0] Total index number is: 28796 (memory: 28796). User sparse processing is: disabled, image size: 720 x 576, area number: 160
2024-10-26T17:37:51.284Z [IMAGETOLED_MNG0] Total index number is: 33120 (memory: 33120). User sparse processing is: disabled, image size: 720 x 576, area number: 160
2024-10-26T17:38:04.825Z [IMAGETOLED_MNG0] Total index number is: 31740 (memory: 31740). User sparse processing is: disabled, image size: 720 x 576, area number: 160
2024-10-26T17:38:11.785Z [IMAGETOLED_MNG0] Total index number is: 33120 (memory: 33120). User sparse processing is: disabled, image size: 720 x 576, area number: 160
2024-10-26T17:38:47.686Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000], [LED0: FPS = 50.00, send = 3000, processed = 3000, dropped = 0]
2024-10-26T17:39:04.546Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 0 frames already dropped
2024-10-26T17:39:05.547Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 1 frames already dropped
2024-10-26T17:39:06.548Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 2 frames already dropped
2024-10-26T17:39:07.550Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 3 frames already dropped
2024-10-26T17:39:08.551Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 4 frames already dropped
2024-10-26T17:39:09.551Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 5 frames already dropped
2024-10-26T17:39:09.551Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:146) Flush was successful
2024-10-26T17:39:41.251Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:164) Serial port is closed: ttyACM0
2024-10-26T17:39:41.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2024-10-26T17:39:41.251Z [LEDDEVICE0_ADALIGHT] Device 'adalight' is disabled due to an error: 'Timeout writing data to ttyACM0'
2024-10-26T17:39:41.251Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T17:39:41.251Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T17:39:47.684Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 2999]
2024-10-26T17:40:04.216Z [WEBSOCKET] (HyperAPI.cpp:957) log streaming activated for client ::ffff:192.168.0.133
```

grssll, 2024-10-26:
```
2024-10-26T17:50:32.091Z [LEDDEVICE0_ADALIGHT] Serial port auto-discovery. Found serial port device: Pico (/dev/ttyACM1 => ttyACM1)
2024-10-26T17:50:32.091Z [LEDDEVICE0_ADALIGHT] Opening UART: ttyACM1
2024-10-26T17:50:32.091Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:209) _serialPort.open(QIODevice::ReadWrite): ttyACM1, Baud rate [2000000]bps
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:214) portName:          ttyACM1
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:215) systemLocation:    /dev/ttyACM1
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:216) description:       Pico
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:217) manufacturer:      Raspberry Pi
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:218) productIdentifier: 0xa
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:219) vendorIdentifier:  0x2e8a
2024-10-26T17:50:32.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:220) serialNumber:      DF62585783634936
2024-10-26T17:50:32.116Z [LEDDEVICE0_ADALIGHT] Detected Rp2040 type board. HyperHDR skips the reset. State: 1, 1
2024-10-26T17:50:32.117Z [LEDDEVICE0_ADALIGHT] ESP sent: 'HyperHDR frames: 0 (FPS), receiv.: 0, good: 0, incompl.: 0, mem1: 0, mem2: 460, heap: 123232'
2024-10-26T17:50:32.118Z [LEDDEVICE0_ADALIGHT] DETECTED DEVICE USING HyperSerialEsp8266/HyperSerialESP32/HyperSerialPico FIRMWARE (Welcome! Awa driver 9.) at 2 msec
2024-10-26T17:50:32.118Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:248) delayAfterConnect for 5000 ms - start
2024-10-26T17:50:34.481Z [SMOOTHING0] Clearing queued colors before: enabling
2024-10-26T17:50:34.481Z [SMOOTHING0] Smoothing queue is cleared
2024-10-26T17:50:34.481Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:247) Enable the device
2024-10-26T17:50:34.481Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:64) Switch on
2024-10-26T17:50:34.481Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:140) Power On
2024-10-26T17:50:34.481Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:367) Starting timer with interval = 20ms
2024-10-26T17:50:34.481Z [COMPONENTCTRL0] LED device: enabled
2024-10-26T17:50:34.482Z [SMOOTHING0] Using alternative smoothing input (0)
2024-10-26T17:50:34.522Z [SMOOTHING0] Using alternative smoothing procedure (0)
2024-10-26T17:50:36.868Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:255) delayAfterConnect for 5000 ms - finished
2024-10-26T17:50:36.869Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:146) Flush was successful
2024-10-26T17:50:36.876Z [LEDDEVICE0_ADALIGHT] Received goodbye: 'HyperHDR frames: 0 (FPS), receiv.: 0, good: 0, incompl.: 0, mem1: 0, mem2: 460, heap: 123232
HyperHDR frames: 51 (FPS), receiv.: 51, good: 51, incompl.: 0, mem1: 0, mem2: 452, heap: 123232
' (191)
2024-10-26T17:50:36.883Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:164) Serial port is closed: ttyACM1
2024-10-26T17:50:39.515Z [SMOOTHING0] Clearing queued colors before: disabling
2024-10-26T17:50:39.515Z [SMOOTHING0] Smoothing queue is cleared
2024-10-26T17:50:39.515Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:297) Disable the device
2024-10-26T17:50:39.515Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2024-10-26T17:50:39.515Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:90) Switch off
2024-10-26T17:50:39.515Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T17:50:42.193Z [SMOOTHING0] Clearing queued colors before: enabling
2024-10-26T17:50:42.194Z [SMOOTHING0] Smoothing queue is cleared
2024-10-26T17:50:42.194Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:247) Enable the device
2024-10-26T17:50:42.255Z [LEDDEVICE0_ADALIGHT] Serial port auto-discovery. Found serial port device: Pico (/dev/ttyACM1 => ttyACM1)
2024-10-26T17:50:42.255Z [LEDDEVICE0_ADALIGHT] Opening UART: ttyACM1
2024-10-26T17:50:42.255Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:209) _serialPort.open(QIODevice::ReadWrite): ttyACM1, Baud rate [2000000]bps
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:214) portName:          ttyACM1
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:215) systemLocation:    /dev/ttyACM1
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:216) description:       Pico
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:217) manufacturer:      Raspberry Pi
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:218) productIdentifier: 0xa
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:219) vendorIdentifier:  0x2e8a
2024-10-26T17:50:42.283Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:220) serialNumber:      DF62585783634936
2024-10-26T17:50:42.286Z [LEDDEVICE0_ADALIGHT] Detected Rp2040 type board. HyperHDR skips the reset. State: 1, 1
2024-10-26T17:50:42.288Z [LEDDEVICE0_ADALIGHT] DETECTED DEVICE USING HyperSerialEsp8266/HyperSerialESP32/HyperSerialPico FIRMWARE (HyperHDR frames: 0 (FPS), receiv.: 0, good: 0, incompl.: 0, mem1: 0, mem2: 452, heap: 123232  Welcome! Awa driver 9.) at 1 msec
2024-10-26T17:50:42.288Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:64) Switch on
2024-10-26T17:50:42.288Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:140) Power On
2024-10-26T17:50:42.288Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:367) Starting timer with interval = 20ms
2024-10-26T17:50:42.288Z [COMPONENTCTRL0] LED device: enabled
2024-10-26T17:50:42.288Z [SMOOTHING0] Using alternative smoothing input (0)
2024-10-26T17:50:42.328Z [SMOOTHING0] Using alternative smoothing procedure (0)
[... cut ...]
2024-10-26T17:57:15.967Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T17:57:47.685Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T17:58:47.670Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 2999, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T17:59:47.671Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
```

grssll, 2024-10-26:
One more log:
```
2024-10-26T19:42:24.106Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 0 frames already dropped
2024-10-26T19:42:25.107Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 1 frames already dropped
2024-10-26T19:42:26.109Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 2 frames already dropped
2024-10-26T19:42:27.110Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 3 frames already dropped
2024-10-26T19:42:28.111Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 4 frames already dropped
2024-10-26T19:42:29.111Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:293) Timeout after 1000ms: 5 frames already dropped
2024-10-26T19:42:29.112Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:146) Flush was successful
2024-10-26T19:43:00.668Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:164) Serial port is closed: ttyACM0
2024-10-26T19:43:00.668Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2024-10-26T19:43:00.668Z [LEDDEVICE0_ADALIGHT] Device 'adalight' is disabled due to an error: 'Timeout writing data to ttyACM0'
2024-10-26T19:43:00.669Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T19:43:12.106Z [WEBSOCKET] (WebSocketClient.cpp:39) New connection from ::ffff:192.168.0.133
2024-10-26T19:43:21.406Z [WEBSOCKET] (HyperAPI.cpp:957) log streaming activated for client ::ffff:192.168.0.133
2024-10-26T19:43:47.683Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 3ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
```

awawa-dev, 2024-10-26:
I cannot reproduce the problem and in your logs I see all kinds of problem with the serial port: where did you try generic firmaware without the problems for a longer time? Beside what your board/system/kernel version and which exactly firmware you used(release secton?)? Adafruit firmware only difference is different GPIO and workaround for boot which is just an increase of timeout when waiting for the board to be ready.

grssll, 2024-10-26:
Problem started after I've changed USB2 grabber to the Elgato HD60x which is USB3.
I've tried all different port and configurations, but Scorpio still freeze and generic works without any problems.

awawa-dev, 2024-10-26:
> Problem started after I've changed USB2 grabber to the Elgato HD60x which is USB3.

well, check your dmesg for any kind of warning and errors then. beside there are a lot of problems with the serial port on the latest kernel. Still I'm thinking that firmware version is not an issues but you can compile firmware for yourself using 'generic' (boot workaround disabled with changed gpio for scorpio).

grssll, 2024-10-26:
6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

awawa-dev, 2024-10-26:
again...check your dmesg when the problem occurs. the circumstances when the problem appeared after changing the grabber to a high-current unit suggest that it lies somewhere else (power supply or its driver). besides, as I wrote, I do not see in the logs a place where the serial port works correctly even on 'generic' firmware

grssll, 2024-10-26:
Have updated the kernel and problem seems to be gone. Will do more testing.
Also, when there is no input and led strip receives
black/power off I still have slight light bleeding from the stirp. Strip is ws2812 COB.
Cheers
```
2024-10-26T21:55:08.193Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000], [LED0: FPS = 50.00, send = 3000, processed = 3000, dropped = 0]
2024-10-26T21:56:08.198Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000], [LED0: FPS = 50.00, send = 3000, processed = 3000, dropped = 0]
2024-10-26T21:57:04.186Z [SMOOTHING0] Clearing queued colors before: disabling
2024-10-26T21:57:04.186Z [SMOOTHING0] Smoothing queue is cleared
2024-10-26T21:57:04.186Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:297) Disable the device
2024-10-26T21:57:04.186Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2024-10-26T21:57:04.186Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:90) Switch off
2024-10-26T21:57:04.186Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:544) Set LED strip to black/power off
2024-10-26T21:57:04.188Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:146) Flush was successful
2024-10-26T21:57:04.200Z [LEDDEVICE0_ADALIGHT] Received goodbye: '
HyperHDR frames: 51 (FPS), receiv.: 50, good: 50, incompl.: 0, mem1: 0, mem2: 448, heap: 123232
' (98)
2024-10-26T21:57:04.204Z [SMOOTHING0] Using alternative smoothing input (0)
2024-10-26T21:57:04.206Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:164) Serial port is closed: ttyACM0
2024-10-26T21:57:04.206Z [COMPONENTCTRL0] LED device: disabled
2024-10-26T21:57:04.206Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 3000]
2024-10-26T21:57:08.201Z [PERFORMANCE] [USB: FPS = 50.00, decoding = 2ms, frames = 3000, invalid = 0, mode = direct], [INSTANCE0: FPS = 50.00, processed = 2999]
```

awawa-dev, 2024-10-26:
> black/power off I still have slight light bleeding from the stirp.

hmm... parasite current? double check your LED strip connection including common ground, maybe something is wrong in the connection diagram or a resistor on the data line may be needed.

awawa-dev, 2024-10-26:
btw do you use a smart socket with a relay to control the LED power supply?

grssll, 2024-10-26:
No smart socket. Separate power supplies for the PI and stirp. Ground connected at RP2040.

awawa-dev, 2024-10-26:
> Also, when there is no input and led strip receives black/power off I still have slight light bleeding from the stirp.

to confirm: LED device is disabled in the remote tab and there is still light bleeding? or for black screen from the grabber there is some backlight?

grssll, 2024-10-26:
Black screen from the grabber and light bleeding. After leds disabled in the remote table they go off completely black.

awawa-dev, 2024-10-26:
ah, this again ;-) ... set minimal backlight to zero in the processing tab.

grssll, 2024-10-26:
You are talking about Backlight threshold from 1% to 0%?

awawa-dev, 2024-10-26:
yes

grssll, 2024-10-26:
👍👍👍Thanks mate. It worked

grssll, 2024-10-27:
For anyone having the same problems please do full RPI upgrade to the latest kernel. It solved freezing issues.
Mod please close and thanks for your help.

closed 2024-10-27 by awawa-dev